PegasusSimulator/PegasusSimulator #1: (Suggestion) 'Getting Started' Video Tutorial

joaolehodey, 2023-02-28:
First of all I am very impressed with this new tool, thank you for sharing your work @MarceloJacinto.

Even tho the Documentation is excellent, I believe that the community would benefit from a "getting started" video tutorial, where installation as well as the basic functionalities are explored and explained. This would serve as a guide for new users, and make it more accessible for the general public.

Thanks again and keep up the good work!!

MarceloJacinto, 2023-02-28:
I think the "getting started" video is a great idea. I will be working on it next!

Thanks for the suggestion

MarceloJacinto, 2023-03-05:
The getting started video was added to the installation tab in the documentation!

closed 2023-03-05 by MarceloJacinto