PegasusSimulator/PegasusSimulator #4: Possibility for MAVROS Backend

mouhyemen, 2023-04-04:
Firstly, fantastic work! This is a great project and will help the Px4 (drones) community working on Isaac Sim.

For offboard control, how easily can we use mavros (let's say with ROS1) for controlling the drone? 

I see that you have a ROS2 backend (with some known issues), but nothing for ROS1.

I am interested in using mavros with ros1 for offboard control of the drone. Would this require me to custom make another backend?

MarceloJacinto, 2023-04-16:
Hey! Sorry for the delay in the response. You do not need to create your custom ROS1 backend to interface with a drone. If you use mavros, the pipeline is as follows:

Pegasus Simulator <-> Mavlink <-> Mavros (or QGroundControl)

On your mavros launch file you must set your IP to be your localhost (if running locally) and the port corresponding to the vehicle's ID. After that, it should behave like a gazebo simulation. However, to get a camera image of a ROS1 topic, you must create your backend! I expect to release an example with a camera and ROS2 soon, so it would not be difficult to replicate using ROS1.

Let me know if this answer was clear.

mouhyemen, 2023-04-16:
Thanks Marcelo. I figured out eventually and already got it working with Mavros.

The camera example would be great. Looking forward to that.

MarceloJacinto, 2023-04-16:
Cool. Ok, so I have just pushed an example using a Camera. It still has a few things that need to be worked out. Please check issue #5

closed 2023-04-16 by MarceloJacinto