PegasusSimulator/PegasusSimulator #5: Question about Sensor support (Especially Camera)

euncheolChoi, 2023-04-14:
First, thanks for sharing your great work.
I have referenced the 'RoadMap' in pegasuss docs for this simulator. According to the docs, there is currently no RGB camera in the list of supported sensors, so I would like to know when the camera sensor will be available in the simulator. 
Also, I was wondering if it is Currently possible to communicate with object via Mavros based on ROS2 in its current state.

Thanks in advance.

MarceloJacinto, 2023-04-16:
Hey, I have been working on publishing the camera feed to ROS 2 topics in my free time. I haven't pushed it yet to a branch, but I might do that soon. Regarding mavros, the short answer is yes! You can use it with the simulation in its current state with mavros - please check this issue: [https://github.com/PegasusSimulator/PegasusSimulator/issues/4](https://github.com/PegasusSimulator/PegasusSimulator/issues/4)

petergerten, 2023-11-04:
As there seem to be no new development on the camera feature in recent month, I wonder if someone can say something about its maturity. Can a RGB camera now reliably be used?

closed 2024-04-13 by MarceloJacinto