PegasusSimulator/PegasusSimulator #12: Add Support for ArduPilot Backend

TomerTip, 2023-06-22:
Additional backend support for ArduPilot SITL, which is another trusted, versatile, and open source autopilot system.
- arducopter_launch_tool.py
- mavlink_backend.py: Remove hardcoded px4 references
- Be able to choose the wanted autopilot

MarceloJacinto, 2023-06-26:
Hi, do you have the suggested file arducopter_launch_tool.py that you can commit through a pull request? If so, that would be a really nice addition.

TomerTip, 2023-06-26:
Hello Marcelo,
I am willing to implement this feature, and submit a pull request, hopefully in this following month 😄

Dat-Bois, 2023-09-22:
Hi @TomerTip, also super interested in adding this as a feature, any progress? I am willing to support as needed. 

Also will this backend also be using mavlink or do you think a mavros implementation is reasonable? This could allow both PX4 and ardupilot to work with the same interface (with some topic alterations ofc).

TomerTip, 2023-09-23:
> Hi @TomerTip, also super interested in adding this as a feature, any progress? I am willing to support as needed.
>
> Also will this backend also be using mavlink or do you think a mavros [...]

Hi,
Sorry guys, no progress yet from my side (life got in the way)...

Please feel free to start the development and hopefully we can join together to achieve it.

My opinion is we should first support Ardupilot backend with plain Mavlink.
Adding mavros support afterwards should be more trivial.

Dat-Bois, 2023-09-23:
Ok got it, no worries :) 

Instead of me blindly looking for which files to change/add for Ardupilot support, can you point me towards where the changes or additions need to be made? You don't need to necessarily tell me what to change, but just the files you think I should start taking a look at and I can ask you questions as needed. 

My guess is to start from [mavlink_backend.py](https://github.com/PegasusSimulator/PegasusSimulator/blob/main/extensions/pegasus.simulator/pegasus/simulator/logic/backends/mavlink_backend.py)

MarceloJacinto, 2024-02-03:
I will close this issue, because I don't have plans to support Ardupilot, and no pull request was opened adding support for this feature by an external contributor.

closed 2024-02-03 by MarceloJacinto