PegasusSimulator/PegasusSimulator #16: Can't see Streaming Backend field in the Pegasus Simulator tab on the Isaac-sim-2023.1.0

maponarooo, 2023-11-08:
First of all, I would like to express my infinite appreciation for this project.
This project solved many of the test problems of PX4 and Isaac-sim that I have been curious about.
I found the following problems while installing the version of Isaac-sim-2023.1.0 according to the guide document.
I can't see Streaming Backend field in the Pegasus Simulator tab on the Isaac-sim-2023.1.0.
This is my screen capture.
I want to know what the problem is.
[image]

I look forward to your kind reply.

Herwich012, 2023-11-13:
Have you tried it with Isaac Sim version 2022.2.0? Even with Isaac Sim version 2022.2.1 some features were not properly working for me. Therefore I have been sticking with 2022.2.0 for now.

MarceloJacinto, 2024-02-03:
Today I've updated most features to work with the newer version 2023.1.1. I disabled the ROS2 streaming backend via GUI, because I was not liking the way it was behaving. I should fix that soon. I've also added scroll bars to fix the viewing problem for smaller screens.

closed 2024-02-03 by MarceloJacinto