PegasusSimulator/PegasusSimulator #18: Failed to load the scene with v1.0.0, ubuntu 22, ros2 humble

otemiz, 2024-01-06:
I followed the installation instructions with Ubuntu 22.04 however when I click load scene I am getting the following output in the terminal and it cannot load.

`
2024-01-06 13:22:34 [186,398ms] [Error] [asyncio] Task exception was never retrieved
future: <Task finished name='Task-333' coro=<PegasusInterface.load_environment_async() done, defined at /home/volo/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/logic/interface/pegasus_interface.py:220> exception=AttributeError("'NoneType' object has no attribute 'warm_start'")>
Traceback (most recent call last):
  File "/home/volo/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/logic/interface/pegasus_interface.py", line 231, in load_environment_async
    await self.world.reset_async()
  File "/home/volo/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/world/world.py", line 520, in reset_async
    await self.reset_async_no_set_up_scene(soft=soft)
  File "/home/volo/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/world/world.py", line 480, in reset_async_no_set_up_scene
    await SimulationContext.reset_async(self, soft=soft)
  File "/home/volo/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 653, in reset_async
    await self.play_async()
  File "/home/volo/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 869, in play_async
    self.get_physics_context().warm_start()
AttributeError: 'NoneType' object has no attribute 'warm_start'
`

Additionally I also cannot proceed with the installation after the pip command. I am getting following error:

# Run the pip command using the built-in python interpreter
ISAACSIM_PYTHON -m pip install --editable pegasus.simulator

`
ERROR: Command errored out with exit status 1:
     command: /home/volo/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/volo/PegasusSimulator/extensions/pegasus.simulator/setup.py'"'"'; __file__='"'"'/home/volo/PegasusSimulator/extensions/pegasus.simulator/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-q0fvb8w7
         cwd: /home/volo/PegasusSimulator/extensions/pegasus.simulator/
    Complete output (1 lines):
    error in pegasus-simulator setup command: 'python_requires' must be a string containing valid version specifiers; Invalid specifier: '>=3.7.*'
    ----------------------------------------
WARNING: Discarding file:///home/volo/PegasusSimulator/extensions/pegasus.simulator. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 21.2.1+nv1; however, version 23.3.2 is available.
You should consider upgrading via the '/home/volo/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/python/bin/python3 -m pip install --upgrade pip' command.
There was an error running python
`

Thanks in advance for the help!

F1rrel, 2024-02-01:
Try changing it [here](https://github.com/PegasusSimulator/PegasusSimulator/blob/8e97926e32f851e0f39a15f766f6ab6c171da6fe/extensions/pegasus.simulator/setup.py#L38) to `python_requires=">=3.7",` and then install the package using `ISAACSIM_PYTHON -m pip install --editable pegasus.simulator`. Worked for me.

MarceloJacinto, 2024-02-03:
Fixed in the last push. Will be closing this issue.

closed 2024-02-03 by MarceloJacinto