PegasusSimulator/PegasusSimulator #19: ModuleNotFoundError: No module named 'toml'

Gampa-Varun, 2024-02-16:
Hello,

My system details are:
Ubuntu 22.04,
ISAACSIM_PYTHON --version is 3.10.13
Isaac sim version is 2023.1.0-hotfix.1

I was following the instructions to install the pegasus extension. (Due to some issue on SIM I was unable to add the path to pegasus extensions folder, as a result, I pasted the pegasus extension folder in one of the existing paths, I was able to see it in extensions and also enable it as per the video)

After enabling the extension, I moved to the pip intsallation:

ISAACSIM_PYTHON -m pip install --editable pegasus.simulator

I get the error:

ISAACSIM_PYTHON -m pip install --editable pegasus.simulator
Obtaining file:///home/gvgampa/Documents/Kit/shared/exts/pegasus.simulator
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build editable did not run successfully.
  │ exit code: 1
  ╰─> [19 lines of output]
      Traceback (most recent call last):
        File "/home/gvgampa/omniverse/Omniverse_lib/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/gvgampa/omniverse/Omniverse_lib/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/gvgampa/omniverse/Omniverse_lib/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 132, in get_requires_for_build_editable
          return hook(config_settings)
        File "/tmp/pip-build-env-rtb6bnmt/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 441, in get_requires_for_build_editable
          return self.get_requires_for_build_wheel(config_settings)
        File "/tmp/pip-build-env-rtb6bnmt/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-rtb6bnmt/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-rtb6bnmt/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 480, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-rtb6bnmt/overlay/lib/python3.10/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 8, in <module>
      ModuleNotFoundError: No module named 'toml'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build editable did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
There was an error running python

Note that  ISAACSIM_PYTHON -m pip install toml
results in 
Requirement already satisfied: toml in /home/gvgampa/omniverse/Omniverse_lib/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (0.10.1)

Furthermore when I launch ISAACSIM_PYTHON 
I am able to import toml and use toml.load function.

If anyone has seen similar issue, please let me know how to resolve if possible. Thanks!

[image]

MarceloJacinto, 2024-02-17:
As a temporary fix, try to run `ISAACSIM_PYTHON -m pip install toml`

Gampa-Varun, 2024-02-20:
Hello, I am very sorry for the late reply. I had already tried that and it reports that requirement is already satisfied:

ISAACSIM_PYTHON -m pip install toml
Requirement already satisfied: toml in /home/gvgampa/omniverse/Omniverse_lib/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.kit.pip_archive/pip_prebundle (0.10.1)

Gampa-Varun, 2024-02-21:
I reinstalled Isaac sim, upgraded to version: isaac_sim-2023.1.1 . This worked! Sorry for the trouble, thanks for the inputs!

closed 2024-02-21 by Gampa-Varun

kevin1zc, 2024-12-10:
This error still persists with Isaac Sim 4.2.0 and Ubuntu 22.04