PegasusSimulator/PegasusSimulator #20: ROS2 Documentation

yashveerjain, 2024-03-29:
First of all Awesome work! I'm trying to use ROS 2 with Isaac Sim and Pegasus, but I couldn't find any getting started guides. While I see Px4 Mavlink integration is the default, are there resources available to implement ROS 2 with Pegasus Simulator?

Thanks in advance.

MarceloJacinto, 2024-04-05:
I still need to write the documentation on that part. I have a script for running with ROS 2 in the examples folder (https://github.com/PegasusSimulator/PegasusSimulator/blob/main/examples/3_ros2_single_vehicle.py) 

No GUI available for that yet, as it had a bug in the previous Isaac Sim version, so I ended up disabling the ROS2 button.  However, the functionality is still there

yashveerjain, 2024-04-06:
First of all great work! and thanks for the update. I am able to run standalone python script, and works great. I'll keep look out for ros2 updates. 
Thanks again.

closed 2024-04-06 by yashveerjain