PegasusSimulator/PegasusSimulator #21: Could not connect with the ROS2

ricardo907, 2024-04-08:
[image]

Hello, When I use the UAV , the streaming Backend is only have PX4? Could you help me to solve it ?

MarceloJacinto, 2024-04-13:
Try to run the following example:
`ISAACSIM_PYTHON examples/3_ros2_single_vehicle.py`

[https://github.com/PegasusSimulator/PegasusSimulator/blob/main/examples/3_ros2_single_vehicle.py](https://github.com/PegasusSimulator/PegasusSimulator/blob/main/examples/3_ros2_single_vehicle.py)

closed 2024-04-13 by MarceloJacinto

261963130, 2025-03-25:
Hello, I'm currently using isaacsim version 4.5, I'm not sure if this version is compatible with your drone simulator or if you will follow up with a compatible version!