PegasusSimulator/PegasusSimulator #22: [BUG] If the tf2_ros Python package is not installed, the installation (or loading?) of the extension would fail

SevenFo, 2024-04-16:
It seems that there's an error related to importing the tf2_ros module in the ros2_backend.py file of the Pegasus Simulator extension. Here's a corrected version of the log:

```
Isaac sim logs "Failed to import python module pegasus.simulator" when I switched on the PegasusSimulator extension in Isaac sim extension manager.

2024-04-16 08:03:49 [158,504ms] [Error] [omni.ext._impl.custom_importer] Failed to import the Python module pegasus.simulator. Error: No module named 'tf2_ros'. Traceback:
Traceback (most recent call last):
  File "/isaac-sim/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "/isaac-sim/kit/python/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/shared/codes/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/__init__.py", line 9, in <module>
    from .extension import Pegasus_SimulatorExtension
  File "/shared/codes/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/extension.py", line 31, in <module>
    from pegasus.simulator.ui.ui_window import WidgetWindow
  File "/shared/codes/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/ui/__init__.py", line 5, in <module>
    from .ui_delegate import UIDelegate
  File "/shared/codes/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/ui/ui_delegate.py", line 22, in <module>
    from pegasus.simulator.logic.backends import MavlinkBackend, MavlinkBackendConfig #, ROS2Backend
  File "/shared/codes/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/logic/backends/__init__.py", line 8, in <module>
    from .ros2_backend import ROS2Backend
  File "/shared/codes/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/logic/backends/ros2_backend.py", line 22, in <module>
    from tf2_ros.static_transform_broadcaster import StaticTransformBroadcaster
ModuleNotFoundError: No module named 'tf2_ros'

2024-04-16 08:03:49 [158,504ms] [Error] [carb.scripting-python.plugin] Exception: Extension Python module 'pegasus.simulator' in '/shared/codes/PegasusSimulator/extensions/pegasus.simulator' failed to load.
```

The import code in 'extensions/pegasus.simulator/pegasus/simulator/logic/backends/ros2_backend.py' 
```
from tf2_ros.static_transform_broadcaster import StaticTransformBroadcaster
from tf2_ros.transform_broadcaster import TransformBroadcaster
```
causes this error, as the ROSBackend class imported in https://github.com/PegasusSimulator/PegasusSimulator/blob/cc3386fb4c5cbd5ebad8ff296b98ef21689d1656/extensions/pegasus.simulator/pegasus/simulator/logic/backends/__init__.py#L8 would import the tf2_ros package.

DavidePisano, 2024-04-17:
i had the same error today ... looking for a solution.

SevenFo, 2024-04-18:
> i had the same error today ... looking for a solution.

You can try checkout this commit (b2cad4ad70680d2fc6156680d7cfcc1f7a7f9703) which can work normally.

MarceloJacinto, 2024-04-19:
Ok, I've made the assumption that most people use the tf package in ROS (for visualization in RVIZ). I should update the ROS2 part of the documentation to include that either as a dependency or just make the tf part optional.

For now, you can just install the tf package as:
`sudo apt-get install ros-humble-rviz2 ros-humble-turtle-tf2-py ros-humble-tf2-ros ros-humble-tf2-tools`

https://docs.ros.org/en/humble/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html

SevenFo, 2024-04-20:
> Ok, I've clearly made the wrong assumption that most people use the tf package in ROS (for visualization in RVIZ). I should update the ROS2 part of the documentation to include that either as a dependency or just make the tf part optional.
> 
> For now, you can install the tf package as: `sudo apt-get install ros-humble-rviz2 ros-humble-tf2-ros ros-humble-tf2-tools`
> 
> https://docs.ros.org/en/humble/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html
> 
> If you are running Ubuntu 22.04 LTS (ROS2 humble), installing the package this way will work just fine.
> 
> If you are running Ubuntu 20.04 LTS (ROS2 foxy), Isaac Sim will NOT use your ROS local install and instead use its internal install. Could this be the culprit of the problems?

And packages `ros-${ROS_DISTRO}-tf2-ros` and `ros-${ROS_DISTRO}-tf2-tools` seem to be accessible only to those who have installed ROS (by configuring the ROS apt source list). It might be inconvenient for those who don't wish to use ROS to install the tf2_ros package.

MarceloJacinto, 2024-04-20:
> @MarceloJacinto #23 OS version: `Ubuntu 22.04 LTS`. I haven't installed ROS2/ROS1 since I run isaac-sim in a Docker container. (I run ROS1 in another Docker container). Additionally, I attempted to utilize isaac-sim's internal backup ROS2 package. The instructions to set it up are as follows:
> 
> ```
> [20.342s] To utilize the internal libraries included with the extension, please set:
> RMW_IMPLEMENTATION=rmw_fastrtps_cpp
> LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/isaac-sim/exts/omni.isaac.ros2_bridge/humble/lib
> Before starting Isaac Sim.
> ```
> 
> However, it appears that this backup ROS2 package also lacks the tf2_ros package.

Ok. This was my suspicion. I need the tf2 package for my personal work, but I will figure out a way to have it optional so that I don’t mess with everyone’s work flow.

SevenFo, 2024-04-20:
> Ok. This was my suspicion. I need the tf2 package for my personal work, but I will figure out a way to have it optional so that I don’t mess with everyone’s work flow.

That sounds like a considerate approach!

Spell1202, 2024-04-29:
> Ok. This was my suspicion. I need the tf2 package for my personal work, but I will figure out a way to have it optional so that I don’t mess with everyone’s work flow.

i am facing the same problem, did you figure the solution out? the only difference is that i am using windows

SevenFo, 2024-04-29:
@Spell1202 
You can wrap the import statement causing the issue with a try-except block. reference: #23

SevenFo, 2024-04-30:
@Spell1202 

 Navigate to your local source code directory (`/…/PegasusSimulator`) and execute the following commands to apply PR #23 to your code

```
curl -L https://github.com/PegasusSimulator/PegasusSimulator/pull/23.patch | git am
```

MarceloJacinto, 2024-08-02:
I've made the TF fully optional in the new commit in the development branch for [Isaac 4.1.0 support](https://github.com/PegasusSimulator/PegasusSimulator/commit/4bb8d8ea492f37dfe6c9147451d256646a7628bd). I will close this issue as this is almost finalized and ready to get merged in main.

closed 2024-08-02 by MarceloJacinto