PegasusSimulator/PegasusSimulator #32: Integration of a Custom Drone

Willyang1379, 2024-05-01:
Hello, firstly I want to thank the creators of the Pegasis Simulator. Currently, I am amidst the process of adding a custom controller; however, the drone is very erratic in nature. My intuition is that the drone's center of mass is off causing the applied force to mess with the drone's movement. Do you have any advice on what steps should be taken to implement a custom drone? For example, checking joint angles, pieces of code to change, etc. Anything would help

Achuthankrishna, 2024-06-21:
Hi @MarceloJacinto , any update on this? I tried my custom URDF and added my model in PX4 as well. My custom model worked perfectly with gz simulator with airframe values similar to iris. How to go about the same using pegasus now?

MarceloJacinto, 2024-11-01:
Hi, sorry I've been a little busy with my PhD. I opened the STL or DAE model file in Blender and exported it to USD (without the props). Then, you must add the revolute joints simulator to the ones found in the Iris vehicle that I load in PegasusSimulator (be careful with the motor ordering and the naming convention, which should be the same as in the Iris drone spawned in the world).

DiegoHerrera1890, 2025-02-21:
Hello guys. 
Thank you so much for sharing this information.
I'm trying to simulate my drone but I'm new to Isaac Sim.
My drone design is SOLIDWORKS assembly. I exported to urdf to use it in a gazebo simulator.

I finally converted it to USD format but I'm running into an issue. When pushing the play button the propeller is not rotating. I do not know if this behavior is correct or not.
In the Iris drone is the same behavior (The difference is that it runs PX4-Autopilot)

When I simulate a cube and the propeller, the propeller rotates over the z axis (RevoluteJoint)

Here's is the picture of my drone config in Isaac Sim, Ignore the error behind because it was solved.
[image]

I would like your advice and any help is welcome.