PegasusSimulator/PegasusSimulator #35: Px4 Commander take off not working

Achuthankrishna, 2024-06-14:
Hi @MarceloJacinto ,
Thanks for the amazing repository contributing to the world of drone simulation. I faced the following issue - after running example/3_ros_single_vehicle.py , I tried to run commander takeoff in px4 shell , and saw that it immediately switched on failsafe and disarmed. I tried with commander arm -f, and it followed the same. The launch threw warnings about the magnteometer , barometer and gps not present though the simulation was running. Is there anything wrong I'm doing? I also tried, ros2 run px4_ros_com offboard_launch , and nothing happened. What installation step am I missing or is the mavlink not connected to PX4 interface ?

MarceloJacinto, 2024-06-14:
Hi @Achuthankrishna,

this is expected because example 3 only uses the ROS 2 backend (no Mavlink backend), therefore the data from the simulator is not being sent through Mavlink and px4 is not being launched. It is being sent to ROS 2 directly. Check [1_px4_single_vehicle.py](https://github.com/PegasusSimulator/PegasusSimulator/blob/main/examples/1_px4_single_vehicle.py), if you want to see the PX4 example.

Note: PX4 is optional in the PegasusSimulator. The idea is that you can use different control/communication backends to choose whether you want to control the vehicle directly from python ([6_paper_results.py](https://github.com/PegasusSimulator/PegasusSimulator/blob/main/examples/6_paper_results.py)), from ROS 2 ([3_ros2_single_vehicle.py](https://github.com/PegasusSimulator/PegasusSimulator/blob/main/examples/3_ros2_single_vehicle.py)) or PX4.

Now, if you want to use PX4 and ROS 2 to control the vehicle in offboard mode, you can also check a separate project that I'm developing at [Pegasus](https://github.com/PegasusResearch/pegasus).

Feel free to join the discord server if more questions like this appear: [https://discord.gg/AjCxw2QUmt](https://discord.gg/AjCxw2QUmt)

Best Marcelo

closed 2024-06-14 by MarceloJacinto

Achuthankrishna, 2024-06-14:
Hi @MarceloJacinto 

I still tried with the 1_px4_single_vehicle.py , and tried to force takeoff, but it threw an error saying 
WARN  [health_and_arming_checks] Preflight Fail: barometer 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Gyro 0 uncalibrated
WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Accel 0 uncalibrated
WARN  [health_and_arming_checks] Preflight Fail: barometer 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Gyro 0 uncalibrated
WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing

And went to disarm. I think this is not the expected behavior. I'm running the microxrceagent parallel to the px4 , is it that isaac sim uses fast_rtps and failing to communicate with px4 ?

Achuthankrishna, 2024-06-14:
More specifically :
pxh> commander takeoff
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_command_ack data writer, topic id: 247
pxh> WARN  [commander] Arming denied: Resolve system health failures first

pxh> commander arm -f
pxh> INFO  [commander] Armed by internal command
INFO  [tone_alarm] arming warning
INFO  [navigator] Using default takeoff altitude: 2.5 m
WARN  [failsafe] Failsafe activated
WARN  [health_and_arming_checks] Preflight Fail: Accel 0 uncalibrated
WARN  [health_and_arming_checks] Preflight Fail: barometer 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Flight termination active
INFO  [tone_alarm] battery warning (fast)
INFO  [commander] Disarmed by lockdown
WARN  [health_and_arming_checks] Preflight Fail: Accel 0 uncalibrated
WARN  [health_and_arming_checks] Preflight Fail: barometer 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
WARN  [health_and_arming_checks] Preflight Fail: Flight termination active
INFO  [tone_alarm] notify neutral