PegasusSimulator/PegasusSimulator #38: ArduPilot Drone takes off in the wrong direction after arming and vibrates during arming

weicong96, 2024-06-18:
I understand this repository does not support ArduPilot and there appears to be nobody currently working on it. 

I had started working on ArduPilot port using mavlink_backend.py but I am facing an issue where, my aircraft will move backwards after arming and takes off pitched backwards. 

I have checked the following:
1. PWM values across all 4 rotors are equal upon takeoff during arming
2. Rotor direction is correct(compared the observed motion with PX4 implementation) 

Initially i thought that,it would be a relatively simple for migrating from HIL messages from PX4 to ArduPilot's JSON SITL Interface, not sure if anyone has any suggestions to try or a direction to explore

MarceloJacinto, 2024-06-20:
Are you sure that the order of the motors in the motor allocation is the same as in PX4?

weicong96, 2024-06-24:
That was my initial theory, here's my procedure when checking the motors allocation.

Procedure of verifying motor allocation:
1) Use PX4 implementation
2) change input_scaling to 0 for all motors
3) set zero_position_armed in mavlink_backend.py to [100, 0, 0, 0] to find direction for first motor, arm and observe direction, then repeat for remaining motors to 100 and leaving others as zero
4) Switch to my ArduCopter implementation
5) Repeat Step 2
6) Repeat Step 3 for all motors

Results that i am getting(with blue props facing front)
PX4 motor orders: top right CW, bottom left CW, top left CCW, bottom right CCW
ArduPilot motor orders: top right CW, bottom left CW, top left CCW , bottom right CCW
I had an additional finding, where the props seems to move in the reverse direction on both PX4 and ArduPilot implemented, ie top right(rotor0) is labelled ccw but spins in clockwise direction.

Would also be interested to find out, the purpose of "rot_direction", does -1 stand for clockwise motion and 1 stand for clockwise motion?

TomerTip, 2024-10-27:
Hi
This behavior is cause as a result of the frame chosen -  `QUADX`. And is fixed when `gazebo_iris` frame is used.
I managed to resolve this issue, and add ArduPilot support (in this PR: https://github.com/PegasusSimulator/PegasusSimulator/pull/54 )
When examining the PWM values received from ArduPilot SITL for frame `QUADX`, I noticed the PWM values for all the rotors are the same and stay consistent, which did not make sense to me, also you can visually watch the drone takes off and crashes the way you described. For frame  `gazebo_iris` frame, SITL calculates the control differently.

closed 2024-11-01 by MarceloJacinto