PegasusSimulator/PegasusSimulator #46: PX4 not loading

alvgaona, 2024-08-20:
Hi,

I tried this simple exercise to load a scene and a vehicle through the PegasusSimulator GUI. Then, I can take off and land from QGroundControl super nicely. However, when I want to save such an environment onto a USD file to open it up another time, I find that PX4 is not starting up anymore when I hit the play button on the UI.

What could this be? Do you need more information from me?

MarceloJacinto, 2024-08-26:
The way this extension is intended to work is to create a multirotor Python object (which will instantiate an iris vehicle in the world). The sensors and communication backend are associated with that multirotor Python object. Suppose you save the world with the iris vehicle already in there. In that case, the multirotor Python object will never get created when you load that USD world (and the Mavlink backend which starts PX4 will not be initialized because it is not there).  

To sum up (and with simpler language), if you save the USD world with the drone in there, you are saving the world with the "drone mesh", but that actual logic that makes it fly is not saved with it.

closed 2024-08-26 by MarceloJacinto

alvgaona, 2024-08-26:
> The way this extension is intended to work is to create a multirotor Python object (which will instantiate an iris vehicle in the world). The sensors and communication backend are associated with [...]

Thank you. I had imagined but had to ask 😄

MarceloJacinto, 2024-08-26:
No problem. Feel free also to ask questions on the [Discord Server](https://discord.gg/AjCxw2QUmt) if needed.