PegasusSimulator/PegasusSimulator #47: GPS coordinate mapping to issac sim coordinates.

shashisingh, 2024-08-21:
First of all, thanks for the work put in to create the simulator. I have been trying to run examples and also trying to understand how different components interact with each other. My use case is to create an outdoor scene and process images are they are being captured by the USD camera in Issac-Sim.  

I sometimes notice errors (drone is stuck on the same frame forever, altitude errors etc.) while flying the simulator through QGroundControl. I am trying to debug these cases. As part of this, I wanted to trace how GPS co-ordinates on QGroundControl map are being mapped to the coordinate space of Issac-Sim scene.  If I want the drone to move towards an object in the scene, how would that be mapped to GPS location on QGroundControl? Thank you for any help.

fyl1991, 2024-08-23:
In this link [link](https://pegasussimulator.github.io/PegasusSimulator/source/features/environments.html#setting-the-map-global-coordinates) for the description of gps information, I hope it will be helpful to you.

MarceloJacinto, 2024-08-26:
I will recommend for you to take a closer look at the [Conference Paper](https://ieeexplore.ieee.org/document/10556959), section III-D (sensor modeling) - the math behind the GPS projection is there. You can also change the coordinates of the center of the projection by following the link suggested by @fyl1991.

The code for the GPS implementation is [here](https://github.com/PegasusSimulator/PegasusSimulator/blob/main/extensions/pegasus.simulator/pegasus/simulator/logic/sensors/gps.py).

Hope this helps.

closed 2024-08-26 by MarceloJacinto