PegasusSimulator/PegasusSimulator #53: AttributeError: 'NoneType' object has no attribute 'initialize_simulation_context_async'

kokoory, 2024-10-05:
Hello, Thanks for your work!

After installing, I could run sim and control using qgc. but I have updated kernel and other things, I am facing an error. 
Nvidia driver and ISAAC version is the same as written on the web.

I could running ISAACSIM, after clicking 'load vehicle' , I've got this error.

Thanks for your help!
------------------------------------------
[Error] [asyncio] Task exception was never retrieved
future: <Task finished name='Task-189' coro=<UIDelegate.on_load_vehicle.<locals>.async_load_vehicle() done, defined at /home/den/Documents/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/ui/ui_delegate.py:177> exception=AttributeError("'NoneType' object has no attribute 'initialize_simulation_context_async'")>
Traceback (most recent call last):
  File "/home/den/Documents/PegasusSimulator/extensions/pegasus.simulator/pegasus/simulator/ui/ui_delegate.py", line 183, in async_load_vehicle
    await self._pegasus_sim.world.initialize_simulation_context_async()
AttributeError: 'NoneType' object has no attribute 'initialize_simulation_context_async'
-----------------------------------------

TannerGilbert, 2024-10-24:
You have to load a scence by clicking "Load Scene" in the scene selector before trying to spawn the vehicle as else the `world` will be `None`.

[image]

MarceloJacinto, 2024-11-01:
Thank you @TannerGilbert for providing the support. This is the correct answer. I will be closing the issue.

closed 2024-11-01 by MarceloJacinto