PegasusSimulator/PegasusSimulator #72: Fixed wing UAV simulation

godwinrayanc, 2025-05-08:
Hello, thank you for the amazing work in the aerial robotics space. I have been working a choosing a simulator for my Fixed wing UAV swarming and found Pegasus. I was wondering is it possible to integrate the Fixed Wing Model and Dynamics in this simulator? And is it straight-forward?

MarceloJacinto, 2025-05-09:
Hi @godwinrayanc, it should be possible to integrate Fixed Wing model dynamics into the PegasusSimulator. I believe the folks at CMU already have some work done for vtols on their fork. You might want to take a look at this link for inspiration:[https://github.com/castacks/PegasusSimulator/blob/main/extensions/pegasus.simulator/pegasus/simulator/logic/vehicles/vtol.py](https://github.com/castacks/PegasusSimulator/blob/main/extensions/pegasus.simulator/pegasus/simulator/logic/vehicles/vtol.py)

godwinrayanc, 2025-05-12:
Hello @MarceloJacinto ,
Thanks for the swift reply! Yes I was able to find this link via discord and it does . I will try it out, do some testing and make a PR if it looks good:)

closed 2025-05-12 by godwinrayanc

KZyred, 2025-05-13:
> Hello [@MarceloJacinto](https://github.com/MarceloJacinto) , Thanks for the swift reply! Yes I was able to find this link via discord and it does . I will try it out, do some testing and make a PR if it looks good:)

I hope you guys can send me the Discord link, as the previous one is no longer accessible.
I'm also trying on implementing CMU's code and contributing to it.
Thank you in advance!

mgrova, 2025-05-13:
@KZyred with the permission of Marcelo, who created the channel, here you have the access invitation: https://discord.gg/276E4krB

Bests,
Marco.