PegasusSimulator/PegasusSimulator #73: Docs: ISAAC SIM installation method

giovannigaiardo, 2025-05-09:
Hello, congrats on the amazing work!

ISAAC SIM is no longer available through the Omniverse Launcher:

[image]

The "Install" option is not showing. Instead, there is an "External Link" button that redirects you to the [Download ISAAC SIM](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/download.html) web page. The [video tutorial](https://youtu.be/YCp5E8nazag) provided in the installation guide doesn't match this behavior. 

I suggest including the [Workstation Installation](https://docs.isaacsim.omniverse.nvidia.com/4.2.0/installation/install_workstation.html) reference in the installation guide, with slight adaptations to the "Configuring the Environment Variables" section accordingly. 

Thank you!

MarceloJacinto, 2025-05-09:
Hi @giovannigaiardo, thanks for reaching out! You absolutely right. When I finish the article that I'm currently working on, I will try to devote some time to update this project.

giovannigaiardo, 2025-05-10:
> Hi [@giovannigaiardo](https://github.com/giovannigaiardo), thanks for reaching out! You absolutely right. After I finish the article that I'm currently working on, I will try to devote some time to update this project installation guide and IsaacSim version.

Thank you for your fast response! 😄 I took the liberty to open #74 , I think it solves this issue for now.

closed 2025-05-10 by giovannigaiardo