PegasusSimulator/PegasusSimulator #79: performance bottleneck (100 px4 drones)

petergerten, 2025-06-05:
First of all thanks for this great project!

I am trying to simulate 100 px4 drones and wonder where the performance bottleneck is.
The good news: simulation works (due to lockstep)
But is unusable slow.

I am using a 24 core Threadripper + RT3090ti GPU + 128GB Ram.
There seem to be no particular high overall CPU nor GPU load. (top show 5% GPU load over all cored; nvidia-smi ~10% GPU load)

So I wonder if the actual Pegasus system is the bottleneck.
Any suggestion how to debug this / or what can be done to improve this?

b0b023, 2025-06-05:
also see https://github.com/PegasusSimulator/PegasusSimulator/pull/66

petergerten, 2025-06-06:
Hi @MarceloJacinto - thanks.
I will study the code a bit more but it seems that all physics updates are done in serial and could potentially computed in parallel for all drones. Not sure if multi-threading or multiprocessing with queues might be a solution here.

TannerGilbert, 2025-06-07:
I'd also be interested in making the simulation more efficient with multiple agents by using more of the available computation power of the device. For my usecases (2-5 drones) I mainly see a bottleneck with multiple sensors though. With multiple cameras + Lidar I get slow performance of ~20 FPS even with a NVIDIA A6000

varuncraghavendra, 2025-11-13:
Hi @TannerGilbert would love to know how you spawned multi drones in NVIDIA Pegasus and what modifications you did on the PX4 standpoint (In my case im controlling my drones with PX4-ROS2 Bridge over a docker (Nav2)), but not sure where all Ive to make changes to provide setpoints to individual drones (like im currently writing to topics of px4_1/fmu/topics but i want to further write to px4_2/fmu2/topics etc.)

Would really appreciate an answer !

vidneu, 2025-12-01:
Hi @petergerten did you find any solutions or workarounds to improve performance with multiple drones? I am encountering similar problems, with around 12 fps with 10 drones

longzhenren, 2025-12-02:
> Hi [@petergerten](https://github.com/petergerten) did you find any solutions or workarounds to improve performance with multiple drones? I am encountering similar problems, with around 12 fps with 10 drones

May I ask which scenario you are using? I have encountered a similar problem. By reducing the camera resolution to 600x600, I can only launch 3 drones in a downtown blockstage, with only approximately 10fps (using single NVIDIA Titan RTX).

abdul-mannan-khan, 2026-10-06:
Some profiling data for this issue, in case it helps. Setup: Isaac Sim 5.1.0 container, Pegasus v5.1.0,
PX4 v1.14.3 SITL, RTX 4090 with 20 vCPUs, headless with rendering off, physics at 250 Hz on the CPU (the default world settings).

| Vehicles | Time per `world.step()` | Real-time factor |
|---|---|---|
| 1 | 7–8 ms | ~0.5 |
| 10 | 16.3 ms (flat plane) / 17.6 ms (410k-triangle terrain) | ~0.25 |

The cost is roughly 6 ms fixed per step plus about 1 ms per vehicle. GPU utilisation is ~0%, and one Python thread is at
~130% CPU. PX4 lockstep is **not** the bottleneck: setting `enable_lockstep: False` made no measurable difference.

`cProfile` over 1000 steps with one vehicle (cumulative, with profiler overhead):
- `PhysicsContext._step` → `omni.physx fetch_results`: 7.8 ms, of which 2.9 ms is its own time and the rest is
  the Python physics callbacks.
- `Vehicle.update_state`: 1.5 ms, including 0.19 ms for `omni.usd.get_world_transform_matrix(body)`, used only to
  get the attitude. That read also requires PhysX to write every body back to USD at every step.
- `Vehicle.update_sensors`: 1.4 ms, of which magnetometer 0.6, IMU 0.4 and GPS 0.2 (`geo_mag_utils` table lookups,
  `numpy.array` construction).
- `Multirotor.update`: 1.2 ms, and `PX4MavlinkBackend.update`: 0.75 ms.

A cheap win that worked for us (one vehicle 7–8 ms → 5.5 ms per step, ten vehicles 17.6 → 15.7 ms; takeoff and flight behaved the same):
read the attitude from the same dynamic_control pose instead of from USD, and let PhysX write to Fabric instead of USD.

```python
from isaacsim.core.utils.extensions import enable_extension
enable_extension('omni.physx.fabric')
s = carb.settings.get_settings()
s.set_bool('/physics/updateToUsd', False)
s.set_bool('/physics/updateVelocitiesToUsd', False)

# in Vehicle.update_state: pose = dc.get_rigid_body_pose(body); attitude = [pose.r[0], pose.r[1], pose.r[2], pose.r[3]]
```

Questions:
1. Would you accept a PR that reads the attitude from `get_rigid_body_pose().r`, so that `updateToUsd` can be turned
   off? Or is there a reason the attitude comes from USD?
2. For many vehicles, would a batched path be welcome, i.e. one `RigidPrimView`/tensor read for all bodies per step
   instead of per-vehicle dynamic_control calls and per-vehicle callbacks? I'd be happy to help test.
3. Is 250 Hz physics required by the PX4 backend, or is a lower rate (e.g. 200 Hz) expected to work with PX4 SITL?