PegasusSimulator/PegasusSimulator #82: Support for IssacSim 4.5.0

michalpelka, 2025-06-25:
I would like to ask if support for Isaac 4.5.0 is planned or being developed?

Error during activation on Issaac Sim 4.5.0 seems to be missing import.

```
2025-06-25 13:31:40 [267,229ms] [Error] [omni.kit.app._impl] [py stderr]: ImportError: cannot import name '_range_sensor' from 'omni.isaac.range_sensor' (unknown location)
```
Full log available here : https://gist.github.com/michalpelka/106c791ce0225db200c22aa06c4b21ac

MarceloJacinto, 2025-07-05:
I've just updated that branch. It should be fixed now. I guess that only the people API is missing. I will try to fix that as soon as possible and merge with main.

MarceloJacinto, 2025-07-20:
PegasusSimulator for Isaacsim 4.5.0 now available in the main branch.

Closing this issue.