PegasusSimulator/PegasusSimulator #86: Isaac Sim 4.5.0 - crash - [Fatal] [gpu.foundation.plugin] Out of resource descriptors!

Austin-Parks, 2025-07-17:
So I'm running Isaac Sim 4.5.0 and using the Pegasus Simulator dev_4.5.0 branch and initially I had this stinker of an issue that would crash Isaac Sim 30 Seconds or so after initially spawning in the drone:

**[98,492ms] [Fatal] [gpu.foundation.plugin] Out of resource descriptors!**

This would always happen within 30-60 seconds on my system, but _using Grok 4_, I was able to figure out the cause of the crash. It came down to Ubuntu 22.04's default limit on the number of file descriptors allowed per process, which is 1024 by default, but Pegasus Simulator on Isaac Sim 4.5.0 once loaded and running stable is using closer to 1382, so Isaac's Vulkan API back end would crash out on account of that max fd per process setting. Raising that system wide limit solved the   **[Fatal] [gpu.foundation.plugin] Out of resource descriptors!**  error that was crashing the entire simulation on Issac Sim 4.5.0 w/ Pegasus Simulator (dev_4.5.0).

Here's how to raise that limit (All credit to Grok 4):

1. System File Descriptor Limit Too Low: Ubuntu's default is 1024 per process, but Isaac Sim (especially with RTX/Vulkan) can exceed this during multi-GPU detection or resource-heavy sims. Your setup has two GPUs (NVIDIA + AMD iGPU), which triggers shared resource creation (e.g., Vulkan fences/semaphors). NVIDIA's Linux troubleshooting guide explicitly calls this out as a cause for crashes like yours.

Solution:

1. Increase File Descriptor Limits (Recommended First Fix)
This directly addresses the "resource descriptors" error per NVIDIA's guide. It's a common culprit for Vulkan-based apps on Linux with multiple GPUs.

Step 1) Modify the following system file:

    sudo nano /etc/systemd/user.conf
    
        find, uncomment, and set DefaultLimitNOFILE=65535

Step 2) Again, modify the following system file:
    
    sudo nano /etc/security/limits.conf
    
        add the following two lines just above the "# End of file" at the bottom of the file:
        
        * hard nofile 65535
        * soft nofile 65535

After this you will need to reboot your machine, but after that, Pegasus Simulator (dev_4.5.0 branch) should be working fine on Isaac Sim 4.5.0. I've ran over an hour long (20Km flight distance) simulation, and so far, working smooth as butter. Watching the number of file descriptors used by issac sim live while it's running, I haven't seen it go above 1388. Pegasus Simulator may still be leaking file descriptors between spawning the drone and mavlink getting established with QGroundControl, but I'm honestly not sure if that's the case, I know when to admit I'm out of my depth...

For the curious, to get a count of the number of file descriptors used by Isaac Sim 4.5.0 when its running:

sudo lsof -p $(ps aux | grep isaacsim.exp.full.kit | grep -v grep | awk '{print $2}') | wc -l

Ignore the warnings, count number is the last thing printed

If you want to inspect the full live fd list for Issac Sim 4.5.0 while it's running, just remove the wc bit from the command above:

sudo lsof -p $(ps aux | grep isaacsim.exp.full.kit | grep -v grep | awk '{print $2}')

I hope all this actually helps, think this may be the first issue I've ever opened. Wasn't sure what to do / where to put it, but figured I'd share.

MarceloJacinto, 2025-07-18:
I @Austin-Parks, thank you for letting me know about this issue. This is the first time I'm hearing about this, so I think this is something nice to add to the documentation in a new section related to simulation crashes.

Austin-Parks, 2025-07-18:
Yeah Grok pointed to Nvidia who was saying the fd limit per process issue (on Linux) effect multi-gpu setups far more frequently since it doubles the number of file descriptors used by the Vulkan API at any given time. (The integrated Graphics in my AMD consumer grade cpu likely being the culprit in my case)

I suspect that on a true single GPU setup, the fd count would likely fall below the default 1024 limit. Seeing as 1388 is already pretty close. Now that number wouldn’t be halved exactly, just the number used by the Vulkan API, but that may be the reason why you haven’t ran into this issue yourself.

MarceloJacinto, 2025-07-20:
Since this is something I cannot fix directly, I have updated the known issues tab in the documentation [https://pegasussimulator.github.io/PegasusSimulator/source/references/known_issues.html](https://pegasussimulator.github.io/PegasusSimulator/source/references/known_issues.html)

I will be closing this issue.

Thank you for your contribution!

closed 2025-07-20 by MarceloJacinto